pebri86/esplay-retro-emulation #13: settings.h: No such file or directory

wybiral, 2019-11-21:
```
/home/davy/Downloads/esplay-retro-emulation-1.5.1-esplay-micro/esplay-launcher/main/main.c:15:22:
fatal error: settings.h: No such file or directory
compilation terminated.
```

I have esp-idf 3.3 and edited mkrelease.sh to use the proper path for mkfw but my attempts to compile keep being met with this error and I can't seem to find settings.h anywhere in the include path. I've tried releases 1.5.1 and 1.5 with the same error so far.

Am I missing something?

Thanks for the awesome project!

pebri86, 2019-11-21:
Hi, it seems you're missing path to esplay sdk components. try edit the mkrelease.sh and check if path to ESPLAY_SDK environment variable is correct.

wybiral, 2019-11-21:
Thanks, that was the problem. I didn't notice that relative path in the build files, after moving things around to mirror your setup it seems to work fine.

It's building now but I'm getting an error with the mkfw phase that seems to make the resulting firmware fail (it has a bad checksum). I'll try to debug it myself unless you have any ideas.

Should I close this issue since the settings.h was basically my own user error?

```
davy@silver:~/esp/esplay-retro-emulation$ /home/davy/esp/esplay-base-firmware/tools/mkfw/mkfw Retro-Emulation assets/tile.raw 0 16 1048576 launcher esplay-launcher/build/RetroLauncher.bin 0 17 655360 esplay-nofrendo esplay-nofrendo/build/esplay-nofrendo.bin 0 18 655360 esplay-gnuboy esplay-gnuboy/build/esplay-gnuboy.bin 0 19 1376256 esplay-smsplusgx esplay-smsplusgx/build/esplay-smsplusgx.bin
HEADER='ESPLAY_FIRMWARE_V00_01'
FirmwareDescription='Retro-Emulation'
tile: wrote 8256 bytes.
[0] type=0, subtype=16, length=1048576, label=launcher        
part=0, length=965040, data=esplay-launcher/build/RetroLauncher.bin
[1] type=0, subtype=17, length=655360, label=esplay-nofrendo 
part=1, length=527696, data=esplay-nofrendo/build/esplay-nofrendo.bin
[2] type=0, subtype=18, length=655360, label=esplay-gnuboy   
Aborted (core dumped)
```

pebri86, 2019-11-22:
Sure, the issue has been solve then you should close. Regarding the mkfw fail you can open new issue, it help to track the projects issue. 

if i looked at the your console logs, did the gnuboy build correctly? it seems the gnuboy fail when building.

wybiral, 2019-11-22:
I'll try to debug it and make sure gnuboy was building correctly before opening another issue. But I'll close this one now.

Thank you!

closed 2019-11-22 by wybiral