pebri86/esplay-retro-emulation #15: How to restart config the project?

Afantor, 2020-06-05:
This is a great project. but I am a new esp32 learner.Can you guide me how to configure the project？
I have git checkout esp-idf branch to release/v3.3, and install ffmpeg. when I run ./mkrelease.sh, I have to restart config the project. The following is print out:
GENCONFIG
*
* Restart config...
*
*
* Application manager
*
Use time/date stamp for app (APP_COMPILE_TIME_DATE) [Y/n/?] y
Exclude PROJECT_VER from firmware image (APP_EXCLUDE_PROJECT_VER_VAR) [N/y/?] n
Exclude PROJECT_NAME from firmware image (APP_EXCLUDE_PROJECT_NAME_VAR) [N/y/?] n
The length of APP ELF SHA is stored in RAM(chars) (APP_RETRIEVE_LEN_ELF_SHA) [16] (NEW) 
*
* ESP BLE Mesh Support
*
ESP BLE Mesh Support (BLE_MESH) [N/y/?] (NEW)**n**

*
* SPI RAM config
*
Initialize SPI RAM when booting the ESP32 (SPIRAM_BOOT_INIT) [Y/n/?] y
  Ignore PSRAM when not found (SPIRAM_IGNORE_NOTFOUND) [Y/n/?] y
SPI RAM access method
> 1. Integrate RAM into ESP32 memory map (SPIRAM_USE_MEMMAP)
  2. Make RAM allocatable using heap_caps_malloc(..., MALLOC_CAP_SPIRAM) (SPIRAM_USE_CAPS_ALLOC)
  3. Make RAM allocatable using malloc() as well (SPIRAM_USE_MALLOC)
choice[1-3?]: 1
Type of SPI RAM chip in use
> 1. Auto-detect (SPIRAM_TYPE_AUTO)
  2. ESP-PSRAM32 or IS25WP032 (SPIRAM_TYPE_ESPPSRAM32)
  3. ESP-PSRAM64 or LY68L6400 (SPIRAM_TYPE_ESPPSRAM64)
choice[1-3]: 1
Set RAM clock speed
  1. 40MHz clock speed (SPIRAM_SPEED_40M)
> 2. 80MHz clock speed (SPIRAM_SPEED_80M)
choice[1-2?]: 2
Run memory test on SPI RAM initialization (SPIRAM_MEMTEST) [Y/n/?] y
Enable workaround for bug in SPI RAM cache for Rev1 ESP32s (SPIRAM_CACHE_WORKAROUND) [Y/n/?] y
Enable bank switching for >4MiB external RAM (SPIRAM_BANKSWITCH_ENABLE) [Y/n/?] y
  Amount of 32K pages to reserve for bank switching (SPIRAM_BANKSWITCH_RESERVE) [8] 8
Allow .bss segment placed in external memory (SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY) [N/y/?] n
SPI host to use for 32MBit PSRAM
  1. HSPI host (SPI2) (SPIRAM_OCCUPY_HSPI_HOST)
> 2. VSPI host (SPI3) (SPIRAM_OCCUPY_VSPI_HOST)
  3. Will not try to use any host, will abort if not able to use the PSRAM (SPIRAM_OCCUPY_NO_HOST)
choice[1-3?]: 2
SPI PSRAM WP(SD3) Pin when customising pins via eFuse (read help) (SPIRAM_SPIWP_SD3_PIN) [7] 7
Enable SPI PSRAM 2T mode (SPIRAM_2T_MODE) [N/y/?] (NEW)y
*
* Wi-Fi
*
Max number of WiFi static RX buffers (ESP32_WIFI_STATIC_RX_BUFFER_NUM) [10] 10
Max number of WiFi dynamic RX buffers (ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM) [32] 32
Type of WiFi TX buffers
> 1. Static (ESP32_WIFI_STATIC_TX_BUFFER)
  2. Dynamic (ESP32_WIFI_DYNAMIC_TX_BUFFER)
choice[1-2?]: 1
Max number of WiFi static TX buffers (ESP32_WIFI_STATIC_TX_BUFFER_NUM) [16] 16
WiFi CSI(Channel State Information) (ESP32_WIFI_CSI_ENABLED) [N/y/?] n
WiFi AMPDU TX (ESP32_WIFI_AMPDU_TX_ENABLED) [Y/n/?] y
  WiFi AMPDU TX BA window size (ESP32_WIFI_TX_BA_WIN) [6] 6
WiFi AMPDU RX (ESP32_WIFI_AMPDU_RX_ENABLED) [Y/n/?] y
  WiFi AMPDU RX BA window size (ESP32_WIFI_RX_BA_WIN) [6] 6
WiFi NVS flash (ESP32_WIFI_NVS_ENABLED) [Y/n/?] y
WiFi Task Core ID
> 1. Core 0 (ESP32_WIFI_TASK_PINNED_TO_CORE_0)
  2. Core 1 (ESP32_WIFI_TASK_PINNED_TO_CORE_1)
choice[1-2?]: 1
Max length of WiFi SoftAP Beacon (ESP32_WIFI_SOFTAP_BEACON_MAX_LEN) [752] 752
WiFi mgmt short buffer number (ESP32_WIFI_MGMT_SBUF_NUM) [32] 32
Enable WiFi debug log (ESP32_WIFI_DEBUG_LOG_ENABLE) [N/y/?] n
WiFi IRAM speed optimization (ESP32_WIFI_IRAM_OPT) [Y/n/?] y
WiFi RX IRAM speed optimization (ESP32_WIFI_RX_IRAM_OPT) [Y/n/?] (NEW) y
*
* LWIP
*
Enable copy between Layer2 and Layer3 packets (L2_TO_L3_COPY) [N/y/?] n
Support receiving and sending ethernet packets with VLAN header (ETHARP_SUPPORT_VLAN) [N/y/?] (NEW)n
Enable LWIP IRAM optimization (LWIP_IRAM_OPTIMIZATION) [N/y/?] n
Max number of open sockets (LWIP_MAX_SOCKETS) [4] 4
Randomize the local port for the first (LWIP_RANDOMIZE_INITIAL_LOCAL_PORTS) [Y/n/?] (NEW) y
 Support LWIP socket select() only (USE_ONLY_LWIP_SELECT) [N/y/?] n
Enable SO_REUSEADDR option (LWIP_SO_REUSE) [N/y/?] n
Enable SO_RCVBUF option (LWIP_SO_RCVBUF) [N/y/?] n
Enable fragment outgoing IP packets (LWIP_IP_FRAG) [N/y/?] n
Enable reassembly incoming fragmented IP packets (LWIP_IP_REASSEMBLY) [N/y/?] n
Enable LWIP statistics (LWIP_STATS) [N/y/?] n
Enable LWIP ARP trust (LWIP_ETHARP_TRUST_IP_MAC) [N/y/?] n
Send gratuitous ARP periodically (ESP_GRATUITOUS_ARP) [Y/n/?] y
  GARP timer interval(seconds) (GARP_TMR_INTERVAL) [60] 60
TCPIP task receive mail box size (TCPIP_RECVMBOX_SIZE) [32] 32
DHCP: Perform ARP check on any offered address (LWIP_DHCP_DOES_ARP_CHECK) [Y/n/?] y
DHCP: Restore last IP obtained from DHCP server (LWIP_DHCP_RESTORE_LAST_IP) [N/y/?] n
Enable IPV6 stateless address autoconfiguration (LWIP_IPV6_AUTOCONFIG) [N/y/?] (NEW) n
TCP/IP Task Stack Size (TCPIP_TASK_STACK_SIZE) [2560] 2560
TCP/IP task affinity
> 1. No affinity (TCPIP_TASK_AFFINITY_NO_AFFINITY)
  2. CPU0 (TCPIP_TASK_AFFINITY_CPU0)
  3. CPU1 (TCPIP_TASK_AFFINITY_CPU1)
choice[1-3?]: 1
Enable LWIP ASSERT checks (LWIP_ESP_LWIP_ASSERT) [Y/n/?] (NEW)y

Finally,not build anything, just jump to Restart config...

Afantor, 2020-06-05:
another error is: assets/Tile.png: No such file or directory
./mkrelease.sh: line 27: /home/esp32/esplay-base-firmware/tools/mkfw/mkfw.exe: No such file or directory

My system is ubuntu16.04.

pebri86, 2020-06-05:
Hi, @Afantor, are you using esplay micro board or generic esp32 module?

1. Clone this [Repo](https://github.com/pebri86/esplay-base-firmware) and read the compile with the esp-idf version mention in the readme. and flash those base firmware onto freshly esp32 module. skip this step if you're using esplay micro board (the bootloader preinstalled on this board).

> another error is: assets/Tile.png: No such file or directory
> ./mkrelease.sh: line 27: /home/esp32/esplay-base-firmware/tools/mkfw/mkfw.exe: No such file or directory
> 
> My system is ubuntu16.04.

2. Convert Tile.xcf on the assets folder to Tile.png using gimp or other tools.
3. You're using Linux so remove .exe on mkfw to use linux binary.
4. uncomment #make menuconfig on the mkrelease.sh and just save and exit if you don't want change anything, it will update the .config file.
5. run ./mkrelease.sh again

Afantor, 2020-06-05:
@pebri86 Thanks for your reply, I will compile the appropriate firmware by myself later.
I am using esplay-micro board now, Is there a guide manual that can directly use your compiled firmware?

pebri86, 2020-11-02:
It's almost 5 month no activity for this issue. I'll close this issue, reopen anytime if you still have the problem

Thanks.

closed 2020-11-02 by pebri86