pebri86/esplay-retro-emulation #16: retro esp32 for esplay micro?

pizvizdik, 2020-06-14:
hi pebri,

id like to ask if it is possible somehow to upload eugenes retro esp32 v.2.5 firmware to original micro hardware?

thanks :)

pebri86, 2020-06-14:
Hi @pizvizdik it's positively possible, i have old version which run well on esplay hw. I think it's easy to port later version, we only need to change some drivers for hardware (e.g some pins assignment and sdcard driver), the limitation only emulators that packs into fw. Esplay only have 4 MB flash, so only 3 emulators could fit into the fw file.

38293312, 2020-07-04:
pebri86 hello! May I use "mp3-player_DAC.fw" to keep displaying please wait! Using the latest version of "esplay-retro-emu.fw" will not work.

pebri86, 2020-07-04:
> pebri86 hello! May I use "mp3-player_DAC.fw" to keep displaying please wait! Using the latest version of "esplay-retro-emu.fw" will not work.I can't work with the latest V2.2 and V2.0 respectively. The emulator can play games normally
> 

Hello to use audio player, make sure you made a folder named audio on the root of sdcard and place the mp3, ogg, wav or flac files on those folder.

38293312, 2020-07-04:
> > pebri86你好！我可以使用“ mp3-player_DAC.fw”继续显示，请稍等！无法使用最新版本的“ esplay-retro-emu.fw”。我不能分别使用最新的V2.2和V2.0。模拟器可以正常玩游戏
> 
> 您好，要在最新的esplay复古emu固件上使用音频播放器，请确保在sdcard的根目录上创建了一个名为audio的文件夹，并将mp3，ogg，wav或flac文件放在这些文件夹中。

Oh. This turned out to be the case, I always thought it was an MP3 folder. Okay, thank you very much, I will try it immediately.

38293312, 2020-07-04:
Hello pebri86! The MP3 playback problem was successfully solved, thank you very much! But I have a new problem now, there is no charging current after the USB is plugged in, it should be almost no charging? Remove the lithium battery, I measured 4.2V with a voltmeter, strange! There is no charging current. (I connected a USB voltage ammeter to my USB charger)

pebri86, 2020-07-04:
There's an issue with current hw board, the charge controller had using the wrong part (LTC4054L), if you have issue with charging then see if you have right part, there's should be LTC4054 without L. If you have the L version the charge current only max 150mA and with current rprog it will limit to 50mA which will very slow. To solve the problem you need swap this ic with exactly LTC4054-4.2 not LTC4054L. See my hardware repository for issue details.

38293312, 2020-07-04:
Hello pebri86, you are really too powerful, you said this question is correct, I use the LTC4054L-4.2 chip. After your explanation, I now use the TP4054 to solve the problem of not being able to charge just now. It is suitable to control the charging power supply at 500MA. Thank you again for sharing such a good project to play for electronics enthusiasts! Hope you have better projects to share with you in the future.

pebri86, 2020-11-02:
I think, the issue has been solve.

closed 2020-11-02 by pebri86