pebri86/esplay-retro-emulation #17: Do you still update this project?

jianjinliao, 2020-10-20:
Hi, pebri.Do you still update this project?

pebri86, 2020-10-20:
Hi, 
Yes there's still much room to improve the projects.
The only problem is time, currently I don't have much free time. But pull request are always welcome to someone who want to improve, or if you have idea or feature request you can tell me, then I will working for it once I have time.

Thank you,

jianjinliao, 2020-10-20:
yes,I like this projects.I have done this project and found that the handle of the button is very bad, so I want to change the button, refer to GB

jianjinliao, 2020-10-20:
Your pcb I don't open,Because I use Altium Designer

pebri86, 2020-10-20:
Just download eagle with free license, then you can edit the design files

jianjinliao, 2020-11-19:
hi，new  issue,I finally reported an error when compiling the build firmware：
./mkrelease.sh: line 27: /home/Administrator/Administrator/esp/esplay-base-firmware-master/tools/mkfw/.mkfw  : No such file or 
 directory

./mkrelease.sh: line 28: esplay-retro-emulation-master: command not found
rm: cannot remove 'esplay-retro-emu.fw': No such file or directory
mv: cannot stat 'firmware.fw': No such file or  directory

but,I don't see (esplay base firmware master- tool-mkfw-------no have mkfw.exe)

Can you give me some guidance?   thank you

jianjinliao, 2020-11-19:
> hi，new issue,I finally reported an error when compiling the build firmware：
> ./mkrelease.sh: line 27: /home/Administrator/Administrator/esp/esplay-base-firmware-master/tools/mkfw/mkfw.exe : No such file or
> directory
> 
> ./mkrelease.sh: line 28: esplay-retro-emulation-master: command not found
> rm: cannot remove 'esplay-retro-emu.fw': No such file or directory
> mv: cannot stat 'firmware.fw': No such file or directory
> 
> but,I don't see (esplay base firmware master- tool-mkfw-------no have mkfw.exe)
> 
> Can you give me some guidance? thank you

Cannot execute binaries

pebri86, 2020-11-20:
Hi please find the mkfw tools on my other repository, esplay-base-firmware.

jianjinliao, 2020-11-20:
yes, i fine it,now esp-idf prompt error:
usage: /home/Administrator/esp/esplay-base-firmware-master/tools/mkfw/mkfw descr                                                                                                                iption tile type subtype length label binary [...]
./mkrelease.sh: line 28: Retro-Emulation: command not found
rm: cannot remove 'esplay-retro-emu.fw': No such file or directory
mv: cannot stat 'firmware.fw': No such file or directory

 

深圳市希尔讯电子有限公司     品质-廖坚进
TEL:15814646445
Email:liaojianjin@foxmail.com
Address:深圳市观澜新田创新工业园A栋4楼

jianjinliao, 2020-11-20:
yes, i fine it,now esp-idf prompt error:
usage: /home/Administrator/esp/esplay-base-firmware-master/tools/mkfw/mkfw descr                                                                                                                iption tile type subtype length label binary [...]
./mkrelease.sh: line 28: Retro-Emulation: command not found
rm: cannot remove 'esplay-retro-emu.fw': No such file or directory
mv: cannot stat 'firmware.fw': No such file or directory

pebri86, 2020-11-20:
Hmm that strange, work fine here. You are compiling using msys env under windows, right? 

Please post your mkrelease.sh here

jianjinliao, 2020-11-20:
#!/bin/sh

#edit your ESP-IDF path here, tested with release/v3.3 branch
export IDF_PATH=E:/esp/msys32/home/Administrator/esp-idf-v3.3

#tune this to match yours
export ESPLAY_SDK=~/esp/esplay-retro-emulation/esplay-sdk

#ffmpeg path
export PATH=/c/ffmpeg/bin:$PATH

cd ~/esp/esplay-retro-emulation/esplay-launcher
#make menuconfig
make -j8
cd ~/esp/esplay-retro-emulation/esplay-gnuboy
#make menuconfig
make -j8
cd ~/esp/esplay-retro-emulation/esplay-nofrendo
#make menuconfig
make -j8
cd ~/esp/esplay-retro-emulation/esplay-smsplusgx
#make menuconfig
make -j8
cd ..

ffmpeg -i assets/Tile.png -f rawvideo -pix_fmt rgb565 assets/tile.raw -y
~/esp/esplay-base-firmware/tools/mkfw/mkfw.exe
Retro-Emulation assets/tile.raw 0 16 1310720 launcher esplay-launcher/build/RetroLauncher.bin 0 17 524288 esplay-nofrendo esplay-nofrendo/build/esplay-nofrendo.bin 0 18 458752 esplay-gnuboy esplay-gnuboy/build/esplay-gnuboy.bin 0 19 1179648 esplay-smsplusgx esplay-smsplusgx/build/esplay-smsplusgx.bin
rm esplay-retro-emu.fw
mv firmware.fw esplay-retro-emu.fw

I suspect my esp-idf v3.34 and I tried to switch ESP IDF 3.2

jianjinliao, 2020-11-20:
No successful build

Administrator@PC-1234 MINGW32 ~/esp/esplay-retro-emulation
#  ~/esp/esplay-base-firmware-master/tools/mkfw/mkfw.exe Retro-Emulation assets/tile.raw 0 16 1310720 launcher esplay-launcher/build/RetroLauncher.bin 0 17 524288 esplay-nofrendo esplay-nofrendo/build/esplay-nofrendo.bin 0 18 458752 esplay-gnuboy esplay-gnuboy/build/esplay-gnuboy.bin 0 19 1179648 esplay-smsplusgx esplay-smsplusgx/build/esplay-smsplusgx.bin rm esplay-retro-emu.fw mv firmware.fw esplay-retro-emu.fw
HEADER='ESPLAY_FIRMWARE_V00_01'
FirmwareDescription='Retro-Emulation'
tile: wrote 8256 bytes.
[0] type=0, subtype=16, length=1310720, label=launcher
Aborted (core dumped)

pebri86, 2020-11-20:
Seems you failed at compiling stage, the error above tells you don't have binary to pack.
check if you have binary every build folder. e.g ~/esp/esplay-retro-emulation/esplay-launcher/build/RetroLauncher.bin

jianjinliao, 2020-11-20:
YES，Failed at compiling stage：
D:/esp/msys32/home/liao/esplay-retro-emulation/esplay-launcher/main/graphics.c:7:23: fatal error: gfxTile.inc: No such file or directory
compilation terminated.
make[1]: *** [/home/liao/esp-idf/make/component_wrapper.mk:290：graphics.o] error 1

pebri86, 2020-11-20:
> YES，Failed at compiling stage：
> D:/esp/msys32/home/liao/esplay-retro-emulation/esplay-launcher/main/graphics.c:7:23: fatal error: gfxTile.inc: No such file or directory
> compilation terminated.
> make[1]: *** [/home/liao/esp-idf/make/component_wrapper.mk:290：graphics.o] error 1

gfxTile.inc generated by ffmpeg at compile. You need to install ffmpeg for your OS then add path or edit path to ffmpeg at mkrelease.sh

jianjinliao, 2020-11-23:
I tried many times, but I couldn't get it gfxTile.inc file，I confirm that ffmpeg is installed.
Is there any way to compile separately gfxTile.inc ?

thank you

jianjinliao, 2020-12-03:
have already been solved gfxTile.inc File problem, closing problem

closed 2020-12-03 by jianjinliao

458941968, 2021-07-19:
> have already been solved gfxTile.inc File problem, closing problem

how do you solved? can you tell me, please. Thank you!

458941968, 2021-07-19:
> > have already been solved gfxTile.inc File problem, closing problem
> 
> how do you solved? can you tell me, please. Thank you!

I solved gfxTile.inc File problem, too.
esplay-launcher\main\component.mk
main.o must change to graphics.o
`graphics.o: gfxTile.inc

gfxTile.inc: $(COMPONENT_PATH)/gfxTile.png
	ffmpeg -i $^ -f rawvideo -pix_fmt rgb565 gfxTile.raw -y
	cat gfxTile.raw | xxd -i > gfxTile.inc`