pebri86/esplay-retro-emulation #18: NES emulator failed to save

mender5, 2020-10-27:
Hi, pebri.
When I save the NES game ,The game will restart and fail to save.  but other simulators can be saved successfully.
I used ESP32-WROVER-IB (8M SRAM) . 
I don't know why.

pebri86, 2020-11-02:
Hi @mender5 
I can't reproduce your issue, I tried the latest release and it's working.
So could you tell me which version did you use and which rom? I tried super mario bros and castlevania rom and it's working.

mender5, 2020-11-02:
Hi @pebri86 
Thank you for your reply.
I use v2.2-esplay-micro and it happened on all nes rom. 
When I save the NES game ,It can be saved normally for the first time. But if I do not exit the NES emulation, Save again will fail and the game will restart.  The SAV file on sd card size becomes zero. Only delete the sav file can  save it again .
ESP32-WROVER-IB  seems to be 8M PSRAM，I don't know if it's the reason.

pebri86, 2020-11-03:
Thanks for your report.
I can reproduce the issue, and there's a memory leak on the libsnss that handling the block memory for emulator's ram. 
Wait for the next release soon.

closed 2020-11-03 by pebri86

pebri86, 2020-11-03:
fixed with v2.3 latest release

bwiklak, 2021-03-11:
Hi, thanks for great project. I still have the issue described above on 2.3 firmware I got from makerfabs few days ago. It's not only on nes games. I even reflashed the firmware and it always fail on save.

pebri86, 2021-03-11:
Hi please, downoad the latest release from github and try again. I tried myself and it's fine

bwiklak, 2021-03-11:
Unfortunately effect is the same. What else can I check? 
My Device options say:
RetroLauncher
v2.3-esplay-micro
IDF v3.3-202-gefdddbb25-dirty

Maybe I could debug it somehow? I'm new to ESP32 but I'd like to play a little bit with this device.

pebri86, 2021-03-12:
Did you have a folder named esplay/data/nes on your sdcard? If not then please create manually, same for all other emulation system

bwiklak, 2021-03-12:
That was it! Thanks for help.
btw, I'd like to play a little bit with this hardware, write some code, maybe a game. I can't figure out if I can connect to the debugging info somehow, do I need a specific additional hardware?

pebri86, 2021-03-12:
Connecting via usb cable will open serial port on the PC, it's either for programming and debugging. You can use esp-idf tools to do both of them without additional hardware

commit f0f84492 (2020-11-03): fixed #18 issue: memory leak NES while load/save