pebri86/esplay-retro-emulation #23: The built firmware cannot be flashed in: Date length error

jianjinliao, 2020-12-05:
I'm sorry to raise the problem again. I successfully compiled the firmware, but bootloader will display 'date long error' when it is flashed in. How can I solve it? Thank you very much!

pebri86, 2020-12-05:
The error means the binary to be flashed not fit in the particular partition, please check the binary size and compare with size in the mkrelease.sh mkfw command ( in bytes size )

jianjinliao, 2020-12-07:
I changed it mkrelease.sh  OK, but there is another problem: partition length alignment error. Is this the problem of data partition table?

pebri86, 2020-12-07:
Yes, there's problem with the partition table. The partition table must be 64 byte align. This partition table created automatically by mkfw when flashing, so the size length must be 64 bytes align.

jianjinliao, 2020-12-08:
Thank you very much for your reply. The problem has been solved. I will compile the firmware suitable for me. 
CLOSE ISSUE

closed 2020-12-08 by jianjinliao