pebri86/esplay-retro-emulation #25: Coleco Roms Crash system

Zach1812, 2021-02-08:
I'm enjoying my ESPlay but have noticed that any Coleco Rom I try to load up crashes the system. Basically it throws a loading loop. The only way for me to get out of the loop is to reset the board with the Menu button on startup. All the ROMS I've tried work with my OpenEmu (Mac) and are in .col format. 
NES, Gameboy, Gameboy Color, and Game Gear all run great. SMS is a little slow, especially with Sonic the Hedgehog. I finally picked up a speaker and finished printing a case for it.

Now if I could get Coleco running, everything would be peachy. Any thoughts?

pebri86, 2021-02-08:
Hi @Zach1812 

Thanks for reporting, did you run the latest release or fw version ship by makerfabs?

Zach1812, 2021-02-08:
Hi @pebri86 

I'm using the fw version by makerfabs. Ver: 20190805-890e20280e.

I'd love to update to the newest fw if at all possible. However, my microusb port broke on the first night (trying to charge) and despite my attempts to fix it I was unable to. I assume the usb port is the only way to update the fw. (For charging purposes, I'm opening my case, and attaching the battery to an adafruit microlipo charger.

pebri86, 2021-02-09:
Please download the latest .fw file at the release page. 
You can update the firmware via sdcard. Place the fw file under /esplay/firmware folder on the root sdcard then enter bootloader mode by press menu while powering on, wait til bootloader screen appear. Just select fw file to be flash by A button then start button to begin flashing

Zach1812, 2021-02-09:
Followed your instructions and still have a boot fail on all colecovision games. Menu says...

RetroLauncher
v2.3-esplay-micro
IDF v3.3-202-gefdddbb25-dirty

pebri86, 2021-02-09:
It's strange work well on mine, Ah did you have a BIOS file? Coleco emulation needs bios file in order to run. Place the bios with named BIOS.col in /roms/col folder at the root of sdcard

Zach1812, 2021-02-09:
That fixed it. I had the BIOS in there but it was incorrectly named. Thanks for the help! Btw what are the button controls for Coleco?

closed 2021-02-10 by Zach1812