pebri86/esplay-retro-emulation #27: Header Mismatch Error

basketballsteve, 2021-04-25:
I have a new ESPlay micro received this week (red case and button caps). I am trying to update the firmware with the Retro ESP32.fw file. I have copied the file into the firmware folder by windows 10. I place the sd card into the ESPlay Micro and boot to the bootloader screen. I see 3 choices in the bootloader screen

esplay-retro-emu
mp3-player_dac
Retro ESP32

If I select the Retro ESP32  i am greeted by  screen that says

01 Retro ESP32

HEADER MATCH ERROR

{START}

{B}  Cancel

If I choose start it will attempt to write by continuously report write error and at the conclusion I will be returned to the bootloader menu with the same choices. I can return the device to the original state by selecting the first choice

esplay--retro-emu

And then i am back to square 1.

I have tried the newest Retro ESP32.fw file as well as older versions of the file but all give the same HEADER MISMATCH error.

monsieurPiGG, 2022-11-08:
Hi, 
You should use retroESP32 from this repository : 
https://github.com/pebri86/esplay-micro-firmware-collections

basketballsteve, 2022-11-08:
Thank you for your response.

nikvostro, 2023-11-02:
Hello there!
@pebri86 I'd like to ask vice versa: could you please guide on how to adapt (compile with some cfg modifications?) Odroid .fw for use with ESPlay HW?

pebri86, 2023-11-04:
hi, please check esplay base firmware repo, there is available fw pack tool for esplay, and to adapt odroid fw to esplay hw you need to mapping pin e.g for input and the sdcard

closed 2024-12-22 by pebri86