pebri86/esplay-retro-emulation #29: Can't read games from SD card

jd5000, 2021-09-10:
I DIY an Esplay Micro compatible device using ESP32 WROVER, flashed esplay base firmware to it, and then successfully flashed "esplay-retro-emu.fw" from SD Card. However once it rebooted into Esplay Retro main interface, it could not find the *.nes games under the \roms\nes folder, always showed "0 games available", do you know what's the issue? Based on my understanding, the base firmware could read and flashed "esplay-retro-emu.fw" file from SD Card, it means the SD Card had been detected successfully. I have made sure the folders on SD Card were unzip from your "skeleton.zip" file.

pebri86, 2021-09-10:
Hello, could you try to debug using serial, if there's any useful information

jd5000, 2021-09-11:
Thanks for reply, following is the serial output during boot up:

ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6736
ho 0 tail 12 room 4
load:0x40078000,len:8900
load:0x40080000,len:7392
entry 0x4008039c
[0;32mI (30) boot: ESP-IDF v3.1-odroid-1-g08f2c8f26 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 11:33:25[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) qio_mode: Enabling default flash chip QIO[0m
[0;32mI (41) boot: SPI Speed      : 80MHz[0m
[0;32mI (46) boot: SPI Mode       : QIO[0m
[0;32mI (50) boot: SPI Flash Size : 16MB[0m
[0;32mI (54) boot: Partition Table:[0m
[0;32mI (57) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (87) boot:  3 firmware         factory app      00 00 00010000 00040000[0m
[0;32mI (94) boot:  4 launcher         OTA app          00 10 00050000 00140000[0m
[0;32mI (102) boot:  5 esplay-nofrendo  OTA app          00 11 00190000 00080000[0m
[0;32mI (109) boot:  6 esplay-gnuboy    OTA app          00 12 00210000 00070000[0m
[0;32mI (117) boot:  7 esplay-smsplusgx OTA app          00 13 00280000 00120000[0m
[0;32mI (125) boot: End of partition table[0m
[0;32mI (129) esp_image: segment 0: paddr=0x00050020 vaddr=0x3f400020 size=0x7ac54 (502868) map[0m
[0;32mI (271) esp_image: segment 1: paddr=0x000cac7c vaddr=0x3ffb0000 size=0x031d4 ( 12756) load[0m
[0;32mI (275) esp_image: segment 2: paddr=0x000cde58 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (278) esp_image: segment 3: paddr=0x000ce260 vaddr=0x40080400 size=0x01db0 (  7600) load[0m
[0;32mI (289) esp_image: segment 4: paddr=0x000d0018 vaddr=0x400d0018 size=0xa7718 (685848) map[0m
[0;32mI (476) esp_image: segment 5: paddr=0x00177738 vaddr=0x400821b0 size=0x1401c ( 81948) load[0m
[0;32mI (502) esp_image: segment 6: paddr=0x0018b75c vaddr=0x400c0000 size=0x0006c (   108) load[0m
[0;32mI (515) boot: Loaded app from partition at offset 0x50000[0m
[0;32mI (515) boot: Disabling RNG early entropy source...[0m
audio_init: sample_rate=44100
input_gamepad_init done.
ILI9341 initialization.
Enable backlight.
Backlight initialization done.
ADC: Characterized using eFuse Vref
[0;31mE (11685) sdmmc_req: sdmmc_host_wait_for_event returned 0x107[0m
[0;31mE (11685) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107[0m
sdcard_open: esp_vfs_fat_sdmmc_mount failed (263)

Ready, AP on channel 5
opendir failed.

jd5000, 2021-09-11:
I tried several regular SD Cards (9 pins), get the same result. My connection is:
Pin3 CMD I/O -> ESP32 GPIO15
Pin4 -> GND
Pin5 -> 3V3
Pin6 CLK -> ESP32 GPIO14
Pin7 -> GND
Pin8 DAT0 -> ESP32 GPIO2

Pin3, Pin6, Pin8 are pulled up to 3V3 by 10K resistors.

Is it possible to add the same SD Card support as base firmware to the "esplay-retro-emu.fw"? 
Thank you very much!

pebri86, 2021-09-11:
Hmm your connection seems okay, did u compile yourself the fw file?

jd5000, 2021-09-11:
No, I used your compiled "esplay-retro-emu.fw" file.

pebri86, 2021-09-12:
It's strange if esplay base firmware works then it should be work for emulator also. Press menu button while powering on should be back to bootloader, then try reflashing the fw file again

jd5000, 2021-09-12:
Yes, I refreshed several times, it could always see SD card in base firmware and flashed  "esplay-retro-emu.fw" file from SD card successfully, but once booted into esplay retro main screen, it could not see the SD card anymore. It's weird. Not sure if the base firmware and esplay retro use different SD card drivers?

pebri86, 2021-09-12:
Hi, yes the only different in the driver is in emulator a use highspeed freq mode setting. could you compile yourself esplay emulator from source? try disabling highspeed mode in sdcard driver, cause to use highspeed mode you have to make sure a good connection.

jd5000, 2021-09-12:
Thank you for the reply. Can you shed me some light on how to disable high speed mode in sdcard driver? Since I am still pretty new to ESP-IDF, if it does not trouble you too much, could you help to compile a firmware with high speed mode disabled? I will try it and report to you the result.

pebri86, 2021-09-15:
Sorry i broke my esp environtment, i will try next weekend. If you want to try yourself, the driver is in esplay-sdk folder.

jd5000, 2021-09-15:
Thank you very much! I tried couple times recently, but could not compile successfully, looked like the ESP-IDF V3. 3 I installed was not the right version. If you can compile the firmware for me, that wiil be great!

closed 2024-12-22 by pebri86