pebri86/esplay-retro-emulation #30: Can you update to the latest IDF

FantasyGmm, 2022-06-08:
`/home/gm/esplay-retro-emulation/esplay-launcher/components/acodecs/src/xmplite/loaders/xm_load.c: In function 'xm_load':
/home/gm/esplay-retro-emulation/esplay-launcher/components/acodecs/src/xmplite/loaders/xm_load.c:645:38: error: 'snprintf' output truncated before the last format character [-Werror=format-truncation=]
  snprintf(tracker_name, 20, "%-20.20s", xfh.tracker);
                                      ^
/home/gm/esplay-retro-emulation/esplay-launcher/components/acodecs/src/xmplite/loaders/xm_load.c:645:2: note: 'snprintf' output 21 bytes into a destination of size 20
  snprintf(tracker_name, 20, "%-20.20s", xfh.tracker);
  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [/home/gm/esp/esp-idf/make/component_wrapper.mk:298: src/xmplite/loaders/xm_load.o] Error 1
make: *** [/home/gm/esp/esp-idf/make/project.mk:649: component-acodecs-build] Error 2
make: *** Waiting for unfinished jobs....`
When I manually compile esplay launcher, the above error will appear. Other things are compiled normally. 
I use idf4.4.1. I don't know why this error occurs
I can't compile others, for example, gnuboy lacks display h. Nofrendo is missing settings h

pebri86, 2022-06-10:
Actually i have it in my local repo, but it's very experimental, also the current code only support idf v4.2 not the latest one 4.4 because conflict of DMA between sdcard sdmmc and i2s driver

pebri86, 2022-06-10:
> `/home/gm/esplay-retro-emulation/esplay-launcher/components/acodecs/src/xmplite/loaders/xm_load.c: In function 'xm_load': [...]

to suppress warning threated error above try using flag Wno-format-truncation when compile

FantasyGmm, 2022-06-13:
Can you create a branch to upload the code? Maybe I can make it run on S3

pebri86, 2022-06-14:
> I'm trying to port it to S3. The base firmware can run, butthe simulator can't be compiled

Can S3 run on idf 4.2? i tried idf 4.4 but it still incomplete,  the source compiled successfully but i2s and sdcard in sdmmc mode can't run together

FantasyGmm, 2022-06-14:
At present, the only release of S3 is 4.4.1, but the SDMMC of the base firmware can not be attached anyway. It has been timeout all the time. It is useless to replace the TF Card

FantasyGmm, 2022-06-14:
I found the reason. SDMMC forgot to be grounded. Now it is normal. I asked my friend. He said that I2S and SDIO of S3 can work together

FantasyGmm, 2022-06-14:
Thank you very much. some files is missing. I don't know where to get these files，If you have time in the near future, can you fix the this project？

FantasyGmm, 2022-06-19:
Excuse me, can you compile under idf4.4.1 now？

pebri86, 2022-06-20:
> Excuse me, can you compile under idf4.4.1 now?

I'm sorry, i need to setup again my environment. i just reinstall the os of my PC. i'll make a separate branch for that.##

FantasyGmm, 2022-07-02:
Can I compile it now? Maybe you can upload a complete project (some header files are missing now), and then I'll find a way to do it myself

FantasyGmm, 2022-08-13:
I recently migrated the master to the new version of IDF and rewritten the build system to use cmake recommended by ESP IDF
 I will weld a new board to test it tomorrow
https://github.com/FantasyGmm/esplay-retro-emulation

FantasyGmm, 2022-08-14:
Now I have three independent firmware, when I use mkfw to combine them FW firmware swipes under BL, the 0th firmware will be written normally, and the subsequent firmware will have erase error. After that, the firmware cannot be started normally after the writing is completed

pebri86, 2022-08-14:
> Now I have three independent firmware, when I use mkfw to combine them FW firmware swipes under BL, the 0th firmware will be written normally, and the subsequent firmware will have erase error. After that, the firmware cannot be started normally after the writing is completed

What the error message when writing the firmware? is it length error? Did you check the binary size? is the partition for each binary size has enough space?

FantasyGmm, 2022-08-15:
`menu_main: fileName='/sd/esplay/firmware/ESPLAY~1.FW'
flash_firmware: HEAP=0x0081667b
Opening file '/sd/esplay/firmware/ESPLAY~1.FW'.
Header OK: 'ESPLAY_FIRMWARE_V00_01'
FirmwareDescription='Retro-Emulation'
flash_firmware: expected_checksum=0x957b4974
flash_firmware: checksum=0x957b4974
flash_firmware: FLASH_START_ADDRESS=0x00070000
Erasing ... (0)
Writing (0) - 00000000
Writing (0) - 0x001000
Writing (0) - 0x002000
Writing (0) - 0x003000
Writing (0) - 0x004000
Writing (0) - 0x005000
....
Writing (0) - 0x13b000
Writing (0) - 0x13c000
OK: [0] Length=0x13cc60
part 0:
	magic=0x50aa
	type=0x01
	subtype=0x02
	[pos.offset=0x00009000, pos.size=0x00004000]
	label='nvs             '
	flags=0000000000

part 1:
	magic=0x50aa
	type=0x01
	subtype=0000
	[pos.offset=0x0000d000, pos.size=0x00002000]
	label='otadata         '
	flags=0000000000

part 2:
	magic=0x50aa
	type=0x01
	subtype=0x01
	[pos.offset=0x0000f000, pos.size=0x00001000]
	label='phy_init        '
	flags=0000000000

part 3:
	magic=0x50aa
	type=0000
	subtype=0000
	[pos.offset=0x00010000, pos.size=0x00060000]
	label='firmware        '
	flags=0000000000

part 4:
	magic=0xebeb
	type=0xff
	subtype=0xff
	[pos.offset=0xffffffff, pos.size=0xffffffff]
	label='????_???Dz???
                            ?qJ1?n?...."(label is broken,too many random char)
'
	flags=0x6eb2314a

flash_firmware: current flash addr 1ACC60,1A0000Erasing ... (1)
spi_flash_erase_range failed. eraseBlocks=157
Writing (1) - 00000000
Writing (1) - 0x001000
....
`
I used this function `print_partitions();` to print the partition table after writing. Only the first firmware was successfully written to the partition. However, it is obvious that the written Part4 partition information is incorrect. I don't know why. I fork the base and the warehouse and submit the modified code. Can you check it

FantasyGmm, 2022-08-15:
I forgot to say that I use ESP32S3N8R8, with 8MB flash and 8MB PSRAM

pebri86, 2022-08-15:
> ```
> menu_main: fileName=/sd/esplay/firmware/ESPLAY.FW
> flash_firmware: HEAP=0x0081667b
> Opening file /sd/esplay/firmware/ESPLAY.FW.
> Header OK: ESPLAY_FIRMWARE_V00_01
> FirmwareDescription=Retro-Emulation
> flash_firmware: expected_checksum=0x957b4974
> flash_firmware: checksum=0x957b4974
> flash_firmware: FLASH_START_ADDRESS=0x00070000
> Erasing ... (0)
> Writing (0) - 00000000
> Writing (0) - 0x001000
> Writing (0) - 0x002000
> Writing (0) - 0x003000
> Writing (0) - 0x004000
> Writing (0) - 0x005000
> ....
> Writing (0) - 0x13b000
> Writing (0) - 0x13c000
> OK: [0] Length=0x13cc60
> part 0:
> 	magic=0x50aa
> 	type=0x01
> 	subtype=0x02
> 	[pos.offset=0x00009000, pos.size=0x00004000]
> 	label='nvs             '
> 	flags=0000000000
> 
> part 1:
> 	magic=0x50aa
> 	type=0x01
> 	subtype=0000
> 	[pos.offset=0x0000d000, pos.size=0x00002000]
> 	label='otadata         '
> 	flags=0000000000
> 
> part 2:
> 	magic=0x50aa
> 	type=0x01
> 	subtype=0x01
> 	[pos.offset=0x0000f000, pos.size=0x00001000]
> 	label='phy_init        '
> 	flags=0000000000
> 
> part 3:
> 	magic=0x50aa
> 	type=0000
> 	subtype=0000
> 	[pos.offset=0x00010000, pos.size=0x00060000]
> 	label='firmware        '
> 	flags=0000000000
> 
> part 4:
> 	magic=0xebeb
> 	type=0xff
> 	subtype=0xff
> 	[pos.offset=0xffffffff, pos.size=0xffffffff]
> 	label=(label is broken,too many random char)
> 	flags=0x6eb2314a
> 
> flash_firmware: current flash addr 1ACC60,1A0000Erasing ... (1)
> spi_flash_erase_range failed. eraseBlocks=157
> Writing (1) - 00000000
> Writing (1) - 0x001000
> ....
> ```
> 
> I used this function `print_partitions();` to print the partition table after writing. Only the first firmware was successfully written to the partition. However, it is obvious that the written Part4 partition information is incorrect. I don't know why. I fork the base and the warehouse and submit the modified code. Can you check it

print_partition after writing one part is not make sense since the partition table will be update after all the part of fw written, can you mention the size of all binary file generated after compile? it should be checked and update the mk_release.sh file if the partition reserve is not enough

FantasyGmm, 2022-08-16:
Yes, I found this problem later. The crash occurred because IDF prevented writing to the partition table address,need to modify menuconfig and set the write protection address in spiflash.
 I modified the firmware size every time after compiling. The size is the same as the firmware file size, but the partition misalignment will be promoted during writing. Except for the first one, I can successfully boot the launcher firmware after writing, but I can't start the simulator, The viewing error is that the set OTA boot  API returned 0x1503. It looks like the simulator firmware was written incorrectly

pebri86, 2022-08-16:
> Yes, I found this problem later. The crash occurred because IDF prevented writing to the partition table address,need to modify menuconfig and set the write protection address in spiflash.
>  I [...]

if you modifiy the firmware size then you must make sure the size partition is 64Kb aligned, that's why it is error partition misalignment

FantasyGmm, 2022-08-16:
Does it have to be 64K aligned? I think the official document only needs 4K alignment

FantasyGmm, 2022-08-16:
esp log
```
write_partition_table: startTableEntry=4, startFlashAddress=0x070000
part 0:
	magic=0x50aa
	type=0x01
	subtype=0x02
	[pos.offset=0x00009000, pos.size=0x00004000]
	label='nvs             '
	flags=0000000000

part 1:
	magic=0x50aa
	type=0x01
	subtype=0000
	[pos.offset=0x0000d000, pos.size=0x00002000]
	label='otadata         '
	flags=0000000000

part 2:
	magic=0x50aa
	type=0x01
	subtype=0x01
	[pos.offset=0x0000f000, pos.size=0x00001000]
	label='phy_init        '
	flags=0000000000

part 3:
	magic=0x50aa
	type=0000
	subtype=0000
	[pos.offset=0x00010000, pos.size=0x00060000]
	label='firmware        '
	flags=0000000000

part 4:
	magic=0x50aa
	type=0000
	subtype=0x10
	[pos.offset=0x00070000, pos.size=0x00142510]
	label='esplay-launcher '
	flags=0000000000

part 5:
	magic=0x50aa
	type=0000
	subtype=0x11
	[pos.offset=0x001b3000, pos.size=0x0007ea60]
	label='esplay-gnuboy   '
	flags=0000000000

part 6:
	magic=0x50aa
	type=0000
	subtype=0x12
	[pos.offset=0x00232000, pos.size=0x0007c800]
	label='esplay-nofrendo '
	flags=0000000000

part 7:
	magic=0x50aa
	type=0000
	subtype=0x13
	[pos.offset=0x002af000, pos.size=0x0012ff50]
	label='esplay-smsplusgx'
	flags=0000000000

Booting application.
I (47350) esp_image: segment 0: paddr=00070020 vaddr=3c0a0020 size=7f10ch (520460) map
I (47400) esp_image: segment 1: paddr=000ef134 vaddr=3fc9bf50 size=00ee4h (  3812) 
[... 99 lines cut ...]
I (1818) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1828) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 

AP Disabled, enabled wifi to use File Manager
ui_file_chooser: fileCount = 2
E (6938) esp_image: Segment 0 load address 0x3c040020, doesn't match data 0x001b3020
system_application_set: esp_ota_set_boot_partition failed.
```
mkfw log
```
mkfw: ./mkfw Retro-Emulation assets/tile.raw 0 16 1271056 esplay-launcher esplay-launcher/build/esplay-launcher.bin 0 17 485984 esplay-gnuboy esplay-gnuboy/build/esplay-gnuboy.bin 0 18 509952 esplay-nofrendo esplay-nofrendo/build/esplay-nofrendo.bin 0 19 1195856 esplay-smsplusgx esplay-smsplusgx/build/esplay-smsplusgx.bin 
start mkfw
%  HEADER='ESPLAY_FIRMWARE_V00_01'
%  FirmwareDescription='Retro-Emulation'
%  tile: wrote 8256 bytes.
%  [0] type=0, subtype=16, length=1271056, label=esplay-launcher
%  part=0, length=1271056, data=esplay-launcher/build/esplay-launcher.bin
%  [1] type=0, subtype=17, length=485984, label=esplay-gnuboy
%  part=1, length=485984, data=esplay-gnuboy/build/esplay-gnuboy.bin
%  [2] type=0, subtype=18, length=509952, label=esplay-nofrendo
%  part=2, length=509952, data=esplay-nofrendo/build/esplay-nofrendo.bin
%  [3] type=0, subtype=19, length=1195856, label=esplay-smsplusgx
%  part=3, length=1195856, data=esplay-smsplusgx/build/esplay-smsplusgx.bin
%  main: checksum=0x75f3f1c6
```

pebri86, 2022-08-16:
> Does it have to be 64K aligned? I think the official document only needs 4K alignment

no the 4k is offset of bootloader, the app partition must be 64KB alignment cause the mmu

pebri86, 2022-08-16:
> esp log
> ```
> write_partition_table: startTableEntry=4, startFlashAddress=0x070000
> part 0:
> 	magic=0x50aa
> 	type=0x01
> 	subtype=0x02
> 	[pos.offset=0x00009000, pos.size=0x00004000]
> 	label='nvs [...]

seems it was successfully running

FantasyGmm, 2022-08-17:
yes，launcher can work normally, play music, and select ROM files. When I press a to try to start, it will fail
I modified the base to align the startup address with 64K and the length with 4K. When Python is generated, the length is also changed to 64K. Then it can be started normally, but now there are new problems
```
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379d04
0x40379d04: esp_backtrace_print_from_frame at /Users/fantasy/esp/esp-idf-esplay/components/esp_system/port/arch/xtensa/debug_helpers.c:86

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (29) boot: compile time 23:52:48
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (69) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (77) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  3 firmware         factory app      00 00 00010000 00060000
I (92) boot:  4 esplay-launcher  OTA app          00 10 00070000 00140000
I (99) boot:  5 esplay-gnuboy    OTA app          00 11 001b0000 00080000
I (107) boot:  6 esplay-nofrendo  OTA app          00 12 00230000 00080000
I (114) boot:  7 esplay-smsplusgx OTA app          00 13 002b0000 00130000
I (122) boot: End of partition table
I (126) esp_image: segment 0: paddr=001b0020 vaddr=3c040020 size=278a8h (161960) map
I (163) esp_image: segment 1: paddr=001d78d0 vaddr=3fca07c0 size=03330h ( 13104) load
I (167) esp_image: segment 2: paddr=001dac08 vaddr=40378000 size=05410h ( 21520) load
I (174) esp_image: segment 3: paddr=001e0020 vaddr=42000020 size=33608h (210440) map
I (215) esp_image: segment 4: paddr=00213630 vaddr=4037d410 size=133a4h ( 78756) load
I (233) esp_image: segment 5: paddr=002269dc vaddr=50000000 size=00010h (    16) load
I (233) esp_image: segment 6: paddr=002269f4 vaddr=600fe000 size=00028h (    40) load
I (248) boot: Loaded app from partition at offset 0x1b0000
I (248) boot: Disabling RNG early entropy source...
I (262) opi psram: vendor id : 0x0d (AP)
I (262) opi psram: dev id    : 0x02 (generation 3)
I (262) opi psram: density   : 0x03 (64 Mbit)
I (266) opi psram: good-die  : 0x01 (Pass)
I (271) opi psram: Latency   : 0x01 (Fixed)
I (276) opi psram: VCC       : 0x01 (3V)
I (280) opi psram: SRF       : 0x01 (Fast Refresh)
I (286) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (291) opi psram: BurstLen  : 0x01 (32 Byte)
I (296) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (302) opi psram: DriveStrength: 0x00 (1/1)
W (307) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (318) spiram: Found 64MBit SPI RAM device
I (322) spiram: SPI RAM mode: sram 80m
I (326) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (333) cpu_start: Pro cpu up.
I (337) cpu_start: Starting app cpu, entry point is 0x403796ec
0x403796ec: call_start_cpu0 at /Users/fantasy/esp/esp-idf-esplay/components/esp_system/port/cpu_start.c:640

I (322) cpu_start: App cpu up.
I (639) spiram: SPI SRAM memory test OK
I (647) cpu_start: Pro cpu start user code
I (647) cpu_start: cpu freq: 240000000
I (647) cpu_start: Application information:
I (650) cpu_start: Project name:     esplay-gnuboy
I (655) cpu_start: App version:      v2.3-esplay-micro-17-gf767c89-d
I (662) cpu_start: Compile time:     Aug 17 2022 14:34:34
I (668) cpu_start: ELF file SHA256:  62e7df51db45d843...
I (674) cpu_start: ESP-IDF:          v4.4.2-dirty
I (680) heap_init: Initializing. RAM available for dynamic allocation:
[... 38 lines cut ...]
HEAP:0x807573, FPS:58.997833
HEAP:0x807573, FPS:59.340321
HEAP:0x807573, FPS:59.999428
HEAP:0x807573, FPS:59.007851
HEAP:0x807573, FPS:58.954395
HEAP:0x807573, FPS:59.903038
HEAP:0x807573, FPS:59.015625
HEAP:0x807573, FPS:59.017204
HEAP:0x807573, FPS:59.931744
HEAP:0x807573, FPS:58.925407
HEAP:0x807573, FPS:59.016644
HEAP:0x807573, FPS:59.024139
HEAP:0x807573, FPS:59.915356
HEAP:0x807573, FPS:58.926697
HEAP:0x807573, FPS:59.015736
HEAP:0x807573, FPS:59.994129
HEAP:0x807573, FPS:58.951805
HEAP:0x807573, FPS:58.926460
HEAP:0x807573, FPS:60.005024
HEAP:0x807573, FPS:59.004742
HEAP:0x807573, FPS:58.946957
HEAP:0x807573, FPS:58.975258
HEAP:0x807573, FPS:59.954010
HEAP:0x807573, FPS:59.007336
```

FantasyGmm, 2022-08-17:
```
app_main: Non deep sleep startup
loader: Reading from sdcard.
I (1738) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1738) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1748) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
Initialized. ROM@0x3f800000
loader: rom.name=''
loader: mbc.type=MBC_NONE, mbc.romsize=2 (32K), mbc.ramsize=1 (8K)
app_main: displayBuffer[0]=0x3fcbd398, [1]=0x3fcc879c
sdcard_create_savefile_path: savefile_path='/sd/esplay/data/nes/maliao.nes.sav'
LoadState: fopen /sd/esplay/data/nes/maliao.nes.sav load failed
HEAP:0x8095a7, FPS:68.134201
HEAP:0x8094cf, FPS:58.991680
HEAP:0x8094cf, FPS:58.984657
HEAP:0x8094cf, FPS:59.342155
HEAP:0x8094cf, FPS:59.998333
HEAP:0x8094cf, FPS:59.003036
HEAP:0x8094cf, FPS:58.959671
HEAP:0x8094cf, FPS:59.878895
HEAP:0x8094cf, FPS:59.014492
```
This is because gnuboy tried to read a save file that does not exist at all, so it did not run successfully,

FantasyGmm, 2022-08-17:
I think it may be that the order of esplay gnuboy and esplay nofrendo is reversed. I swapped two partitions. Gnuboy cannot display anything, but OSD menu can be displayed. However, nes simulator still crashes
esplay nofrendo log
```
I (0) cpu_start: App cpu up.
I (633) spiram: SPI SRAM memory test OK
I (641) cpu_start: Pro cpu start user code
I (641) cpu_start: cpu freq: 240000000
I (641) cpu_start: Application information:
I (644) cpu_start: Project name:     esplay-nofrendo
I (650) cpu_start: App version:      v2.3-esplay-micro-14-g0e64cb0-d
I (657) cpu_start: Compile time:     Aug 17 2022 16:17:27
I (663) cpu_start: ELF file SHA256:  9c44a50e0b31971a...
I (669) cpu_start: ESP-IDF:          v4.4.2-dirty
I (674) heap_init: Initializing. RAM available for dynamic allocation:
I (681) heap_init: At 3FCBE490 len 00021B70 (134 KiB): D/IRAM
I (688) heap_init: At 3FCE0000 len 0000EE34 (59 KiB): STACK/DRAM
I (694) heap_init: At 600FE028 len 00001FD8 (7 KiB): RTCRAM
I (701) spiram: Adding pool of 8192K of external SPI memory to heap allocator
I (709) spi_flash: detected chip: generic
I (713) spi_flash: flash io: qio
I (718) sleep: Configure to isolate all GPIO pins in sleep state
I (724) sleep: Enable automatic switching of GPIO sleep configuration
I (731) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (751) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
audio_init: sample_rate=32000
W (761) I2S: APLL not supported on current chip, use I2S_CLK_D2CLK as default clock source
I (771) I2S: DMA Malloc info, datalen=blocksize=2136, dma_buf_count=8
I (771) I2S: I2S0, MCLK output by GPIO0
audio_volume_set: value (20)
A HEAP:0x822827
I (781) gpio: GPIO[40]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (791) gpio: GPIO[41]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (801) gpio: GPIO[42]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
input_gamepad_init done.
ST7789V initialization.
Enable backlight.
Backlight initialization done.
ADC: Characterized using Default Vref
app_main: Unexpected deep sleep reset
osd_getromdata: Reading from sdcard.
I (1671) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1681) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1691) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
app_main: fileSize=49168
NoFrendo start!
audio_volume_set: value (20)
vid_init done
bmp_create: width=256, height=240, overdraw=8
Initialized. ROM@0x3f800000
rom_load: rominfo->filename='rom'
Head: 0x3f800000 (0 0 0 0)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40379bed  PS      : 0x00060230  A0      : 0x803852dc  A1      : 0x3fcc2500  
0x40379bed: heap_caps_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_caps.c:340

A2      : 0x3fcae21c  A3      : 0x3c051e1d  A4      : 0x3fcc2520  A5      : 0x3fcc2500  
A6      : 0x00000008  A7      : 0x00000008  A8      : 0x80379bed  A9      : 0x3fcc24e0  
A10     : 0x00000000  A11     : 0x3fce453c  A12     : 0x3c051e1d  A13     : 0x3fcc2520  
A14     : 0x3fcc2500  A15     : 0x00000008  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace:0x40379bea:0x3fcc25000x403852d9:0x3fcc2520 0x42006cae:0x3fcc2540 0x42007b6a:0x3fcc2560 0x42007dea:0x3fcc2580 0x4200745b:0x3fcc25b0 0x42016492:0x3fcc25f0 0x420073b7:0x3fcc2610 0x42006535:0x3fcc2630 0x42042874:0x3fcc2660 
0x40379bea: heap_caps_free at /Users/fantasy/esp/esp-idf-esplay/components/heap/heap_caps.c:338

0x403852d9: free at /Users/fantasy/esp/esp-idf-esplay/components/newlib/heap.c:39

0x42006cae: bmp_destroy at /Users/fantasy/clionProject/esplay-retro-emulation/esplay-nofrendo/components/nofrendo/bitmap.c:115
[... 113 lines cut ...]
loader: mbc.type=MBC_NONE, mbc.romsize=2 (32K), mbc.ramsize=1 (8K)
app_main: displayBuffer[0]=0x3fcbd398, [1]=0x3fcc879c
sdcard_create_savefile_path: savefile_path='/sd/esplay/data/gb/test.gb.sav'
LoadState: fopen /sd/esplay/data/gb/test.gb.sav load failed
HEAP:0x8095ab, FPS:68.144005
HEAP:0x8094cb, FPS:59.016319
HEAP:0x8094cb, FPS:58.987000
HEAP:0x8094cb, FPS:59.341526
HEAP:0x8094cb, FPS:59.998055
HEAP:0x8094cb, FPS:59.002842
HEAP:0x8094cb, FPS:58.959499
HEAP:0x8094cb, FPS:59.878738
HEAP:0x8094cb, FPS:59.014355
HEAP:0x8094cb, FPS:59.002895
HEAP:0x8094cb, FPS:59.940201
HEAP:0x8094cb, FPS:58.904919
HEAP:0x8094cb, FPS:59.015503
HEAP:0x8094cb, FPS:62.755291
HEAP:0x8094cb, FPS:61.832798
HEAP:0x8094cb, FPS:58.974964
sdcard_create_savefile_path: savefile_path='/sd/esplay/data/gb/test.gb.sav'
savesate: wrote sram addr=0x3fce23d8, size=0x1000, count=1
savesate: wrote sram addr=0x3fce33d8, size=0x1000, count=1
SaveState: savestate OK.
```

pebri86, 2022-08-17:
> I think it may be that the order of esplay gnuboy and esplay nofrendo is reversed. I swapped two partitions. Gnuboy cannot display anything, but OSD menu can be displayed. However, nes simulator [...]

nofrendo crash while loading a rom, did you try another one?

FantasyGmm, 2022-08-18:
I have two NES ROMs and one GB Rom. I tried two NES ROMs, both of which crashed, but they can be used normally on esp32. So is the GB ROM

FantasyGmm, 2022-08-19:
I found that the crash may be caused by the change of the ROM data address. The memory mapping addresses of esp32 and esp32s3 are different. When I change the address to 0x3d000000 or 0x3c000000, it crashes without any trace
I want to ask how to set this address,Esp32s3 is a continuous block of memory, rather than two separate blocks of memory like esp32
esp32 pdf:https://www.espressif.com.cn/sites/default/files/documentation/esp32_datasheet_en.pdf
esp32s3 pdf:https://www.espressif.com.cn/sites/default/files/documentation/esp32-s3_datasheet_en.pdf
```
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (29) boot: compile time 17:41:42
I (29) boot: chip revision: 0
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 8MB
W (45) boot.esp32s3: PRO CPU has been reset by WDT.
W (51) boot.esp32s3: APP CPU has been reset by WDT.
I (57) boot: Enabling RNG early entropy source...
I (62) boot: Partition Table:
I (66) boot: ## Label            Usage          Type ST Offset   Length
I (73) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (80) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (88) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (95) boot:  3 firmware         factory app      00 00 00010000 00060000
I (103) boot:  4 esplay-launcher  OTA app          00 10 00070000 00130000
I (110) boot:  5 esplay-nofrendo  OTA app          00 11 001a0000 00080000
I (118) boot:  6 esplay-gnuboy    OTA app          00 12 00220000 00070000
I (125) boot:  7 esplay-smsplusgx OTA app          00 13 00290000 00130000
I (133) boot: End of partition table
I (137) esp_image: segment 0: paddr=001a0020 vaddr=3c050020 size=1f034h (127028) map
I (169) esp_image: segment 1: paddr=001bf05c vaddr=3fc97100 size=00fbch (  4028) load
I (170) esp_image: segment 2: paddr=001c0020 vaddr=42000020 size=42a30h (272944) map
I (223) esp_image: segment 3: paddr=00202a58 vaddr=3fc980bc size=02358h (  9048) load
I (225) esp_image: segment 4: paddr=00204db8 vaddr=40378000 size=0f100h ( 61696) load
I (243) esp_image: segment 5: paddr=00213ec0 vaddr=50000000 size=00010h (    16) load
I (243) esp_image: segment 6: paddr=00213ed8 vaddr=600fe000 size=00028h (    40) load
I (254) boot: Loaded app from partition at offset 0x1a0000
I (255) boot: Disabling RNG early entropy source...
I (272) opi psram: vendor id : 0x0d (AP)
I (272) opi psram: dev id    : 0x02 (generation 3)
I (272) opi psram: density   : 0x03 (64 Mbit)
I (276) opi psram: good-die  : 0x01 (Pass)
I (281) opi psram: Latency   : 0x01 (Fixed)
I (285) opi psram: VCC       : 0x01 (3V)
I (290) opi psram: SRF       : 0x01 (Fast Refresh)
I (295) opi psram: BurstType : 0x01 (Hybrid Wrap)
I (301) opi psram: BurstLen  : 0x01 (32 Byte)
I (306) opi psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (312) opi psram: DriveStrength: 0x00 (1/1)
W (317) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (328) spiram: Found 64MBit SPI RAM device
I (332) spiram: SPI RAM mode: sram 80m
I (336) spiram: PSRAM initialized, cache is in normal (1-core) mode.
I (343) cpu_start: Pro cpu up.
I (347) cpu_start: Starting app cpu, entry point is 0x403790a4
0x403790a4: call_start_cpu1 at /Users/fantasy/esp/esp-idf-esplay/components/esp_system/port/cpu_start.c:160

I (0) cpu_start: App cpu up.
I (649) spiram: SPI SRAM memory test OK
I (657) cpu_start: Pro cpu start user code
I (657) cpu_start: cpu freq: 240000000
I (657) cpu_start: Application information:
I (660) cpu_start: Project name:     esplay-nofrendo
I (665) cpu_start: App version:      v2.3-esplay-micro-14-g0e64cb0-d
I (672) cpu_start: Compile time:     Aug 17 2022 16:17:27
I (678) cpu_start: ELF file SHA256:  dcc5ff2f8dbd3b26...
I (684) cpu_start: ESP-IDF:          v4.4.2-dirty
I (690) heap_init: Initializing. RAM available for dynamic allocation:
[... 19 lines cut ...]
I (817) gpio: GPIO[42]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
input_gamepad_init done.
ST7789V initialization.
Enable backlight.
Backlight initialization done.
ADC: Characterized using Default Vref
app_main: Unexpected deep sleep reset
osd_getromdata: Reading from sdcard.
I (1687) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1697) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1707) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42002c4b
0x42002c4b: panic_handler at /Users/fantasy/esp/esp-idf-esplay/components/esp_system/port/panic_handler.c:148 (discriminator 3)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1654
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2f88
entry 0x403c9954
I (29) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
```

pebri86, 2022-08-20:
which address did you change?

FantasyGmm, 2022-08-20:
gnuboy loader.c 
```
int rom_load()
{
	byte c, *data, *header;
	int rlen;

//	data = (void*)0x3f800000;//esp32
	data = (void*)0x3D000000;//esp32s3
```
nofrendo main.c
```
//static char* ROM_DATA = (char*)0x3f800000;//esp32
static char* ROM_DATA = (char*)0x3D000000;//esp32s3
```
I only changed the address of these two places, but neither of them can be started to enter the game, they will crash and restart and cannot enter the game

FantasyGmm, 2022-08-20:
the IDF enables the `Make RAM allocatable using malloc() as well` by default,. It is possible that after `malloc` was used elsewhere in the program to allocate memory, part of the PSRAM was used and then overridden when loading the file into PSRAM, causing the crash

pebri86, 2022-08-20:
> the IDF enables the `Make RAM allocatable using malloc() as well` by default,. It is possible that after `malloc` was used elsewhere in the program to allocate memory, part of the PSRAM was used and then overridden when loading the file into PSRAM, causing the crash

you shouldn't malloc, use following options
Select this option by choosing “Integrate RAM into memory map” from CONFIG_SPIRAM_USE

FantasyGmm, 2022-08-20:
I changed it and it still crashed. Let me try to locate the exact line that caused the crash

FantasyGmm, 2022-08-20:
nofrendo log

```
osd_getromdata: Reading from sdcard.
I (1666) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1676) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1686) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
osd_getromdata: copy /sd/roms/nes/hundouluo.nes file to memory 3D000000.
sdcard_copy_file_to_memory: copy addr: 3D000000 ret: 0
```

`sdcard_copy_file_to_memory` part:

```
FILE *f = fopen(path, "rb");
            if (f == NULL)
            {
                printf("sdcard_copy_file_to_memory: fopen failed.\n");
            }
            else
            {
                // copy
                const size_t BLOCK_SIZE = 512;
                while (true)
                {
	                printf("sdcard_copy_file_to_memory: copy addr: %X ret: %X\n",(0x3D000000+ret),ret);
                    __asm__("memw");
                    size_t count = fread((uint8_t *)ptr + ret, 1, BLOCK_SIZE, f);
                    __asm__("memw");
                    ret += count;
                    if (count < BLOCK_SIZE)
                        break;
                }
	            printf("sdcard_copy_file_to_memory: copy done,copy ret:%X",ret);
            }
```
I found that it crashed when copying files to memory, and it crashed at the beginning,

FantasyGmm, 2022-08-21:
I changed the PSRAM memory allocation to the default value, then I used Malloc to allocate memory for ROM, and saved the address to data, so that he would not crash when reading files, but now can not display anything, OSD menu can be used, I opened the emulator debug_trace, he output the executed assembly, Unfortunately I don't know anything about assembly
gnuboy loader.c
```
int rom_load()
{
	byte c, *data, *header;
	int rlen;

//	data = (void*)0x3f800000;
	data = (void*)0x3D000000;
	char *romPath = settings_load_str(SettingRomPath);
	if (!romPath)
	{
		printf("loader: Reading from flash.\n");

		// copy from flash
		spi_flash_mmap_handle_t hrom;

		const esp_partition_t* part = esp_partition_find_first(0x40, 0, NULL);
		if (part == 0)
		{
			printf("esp_partition_find_first failed.\n");
			abort();
		}

		for (size_t offset = 0; offset < 0x400000; offset += 0x100000)
		{
			esp_err_t err = esp_partition_read(part, offset, (void *)(data + offset), 0x100000);
			if (err != ESP_OK)
			{
				printf("esp_partition_read failed. size = %x, offset = %x (%d)\n", part->size, offset, err);
				abort();
			}
		}
	}
	else
	{
		printf("loader: Reading from sdcard,romPath='%s\n", romPath);
		// copy from SD card
		esp_err_t r = sdcard_open(SD_BASE_PATH);
		if (r != ESP_OK)
		{
			//odroid_display_show_sderr(ODROID_SD_ERR_NOCARD);
			abort();
		}

		// load the first 16k
		RomFile = fopen(romPath, "rb");
		if (RomFile == NULL)
		{
			printf("loader: fopen failed.\n");
                        //odroid_display_show_sderr(ODROID_SD_ERR_BADFILE);
			abort();
		}
		//get size
		fseek(RomFile,0L,SEEK_END);
		long romSize = ftell(RomFile);
		rewind(RomFile);
		data = malloc(sizeof(uint8_t)*romSize);
		memset(data,0,sizeof(uint8_t)*romSize);
		if (data == NULL)
		{
			printf("loader: malloc mem fail");
			abort();
		}
		printf("loader: malloc mem size:%ld kb，address:%#x.\n",sizeof(uint8_t)*romSize/1024,(size_t)data);
		printf("loader: free heap: %d kb,%d mb\n",esp_get_free_heap_size()/1024,esp_get_free_heap_size()/1024/1024);
		printf("loader: free internal heap: %d kb\n",esp_get_free_internal_heap_size()/1024);
		// copy
#if 0
[... 1104 lines cut ...]
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=43 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019D FE 94    CP 94h           SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019F 20 FA    JR NZ,-6         SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=00
019B F0 44    LD A,(FF00+44h)  SP=FFFE.0300 BC=0013.00.00 DE=00D8.AF HL=014D.C6 A=16 F=50 --N-C IE=03 IF=03 LCDC=80 STAT=40 LY=16 LYC=
```

FantasyGmm, 2022-08-21:
I don't see any emulator related code in the while loop in Main, so how do I start the emulator, how do I know if the emulator has started successfully, and how do I call the drawing update

pebri86, 2022-08-22:
> I changed the PSRAM memory allocation to the default value, then I used Malloc to allocate memory for ROM, and saved the address to data, so that he would not crash when reading files, but now can [...]

if you want to use malloc options then the following line is not needed
``` c
data = (void*)0x3D000000;
```

and for the entry to the main loop, there's somewhere on main.c or loader on some emulator

FantasyGmm, 2022-08-22:
I commented on that line, but when I changed #if 0 to 1, it would send a crash and there was no way to read the file. Can I read all the content into memory at one time?

pebri86, 2022-08-22:
> I commented on that line, but when I changed #if 0 to 1, it would send a crash and there was no way to read the file. Can I read all the content into memory at one time?

the asm part leave it #if 0,  try change malloc with heap_caps_malloc with SPI RAM flags, see documentation

FantasyGmm, 2022-08-23:
shit, I wrote the code in the middle of the night, and memset wrote it as a file pointer, so it crashed. Now the NES simulator is running well and can run stably 60 frames

FantasyGmm, 2022-08-23:
I used the same method to read gigabytes of ROM, but some ROMS can show the background and then crash, and some don't show anything at all
Those ROMs that crash on startup, if I reset the hardware, they won't crash
I see that you only read the first 16K of ROM, the data after that do not need to read?
```
// load the first 16k
		RomFile = fopen(romPath, "rb");
		if (RomFile == NULL)
		{
			printf("rom_load: fopen failed.\n");
                        //odroid_display_show_sderr(ODROID_SD_ERR_BADFILE);
			abort();
		}
		//get size
		fseek(RomFile,0L,SEEK_END);
		long romSize = ftell(RomFile);
		rewind(RomFile);
		printf("rom_load: file size: %ld kb\n",romSize/1024);
		data = malloc(sizeof(uint8_t)*romSize);
		if (data == NULL)
		{
			printf("rom_load: malloc mem fail");
			abort();
		}
		memset(data,0,sizeof(uint8_t)*romSize);
		printf("rom_load: malloc mem size:%ld kb，address:%#x.\n",sizeof(uint8_t)*romSize/1024,(size_t)data);
		printf("rom_load: free heap: %d kb,%d mb\n",esp_get_free_heap_size()/1024,esp_get_free_heap_size()/1024/1024);
		printf("rom_load: free internal heap: %d kb\n",esp_get_free_internal_heap_size()/1024);
		// copy
#if 0
...
#else
		size_t count = fread((uint8_t*)data, 1, 0x4000, RomFile);
		printf("loader: read count: %d\n",count);
		if (count < 0x4000)
		{
			//odroid_display_show_sderr(ODROID_SD_ERR_BADFILE);
			printf("rom_load: fread failed.\n");
			abort();
		}
#endif
```
gnuboy log this rom doesn't crash
```
rom_load: Reading from sdcard,romPath='/sd/roms/gb/test.gb
I (1724) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1724) gpio: GPIO[14]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (1734) gpio: GPIO[17]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
rom_load: file size: 1024 kb
rom_load: malloc mem size:1024 kb，address:0x3d8073c4.
rom_load: free heap: 7334 kb,7 mb
rom_load: free internal heap: 189 kb
loader: read count: 16384
rom_load: Initialized. ROM@0x3d8073c4
rom_load: rom.name='EXTREME GB'
rom_load: mbc.type=MBC_MBC5, mbc.romsize=64 (1024K), mbc.ramsize=1 (8K)
app_main: displayBuffer[0]=0x3fcbd398, [1]=0x3fcc879c
sdcard_create_savefile_path: savefile_path='/sd/esplay/data/gb/test.gb.sav'
LoadState: fopen /sd/esplay/data/gb/test.gb.sav load failed
HEAP: 7205 KB, FPS:61.747742
HEAP: 7204 KB, FPS:61.289932
HEAP: 7204 KB, FPS:59.141884
HEAP: 7204 KB, FPS:59.738262
HEAP: 7204 KB, FPS:58.835625
```

FantasyGmm, 2022-08-23:
i found out he was on a loop
```
05B1 F0 44    LD A,(FF00+44h)  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B3 FE 91    CP 91h           SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B5 20 FA    JR NZ,-6         SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
```
What could be causing this, a problem reading ROM？
```
0100 00       NOP              SP=FFFE.0000 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0101 C3 50 01 JP 0150h         SP=FFFE.0000 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0150 F3       DI               SP=FFFE.0000 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0151 31 FF DF LD SP,DFFFh      SP=FFFE.0000 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0154 3E 0A    LD A,0Ah         SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0156 EA 00 00 LD (0000h),A     SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=0A F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
0159 3E 01    LD A,01h         SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=0A F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
015B EA 00 21 LD (2100h),A     SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=00 LY=00 LYC=00
015E E0 8F    LD (FF00+8Fh),A  SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
0160 CD 62 17 CALL 1762h       SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
1762 AF       XOR A            SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=B0 -Z-HC IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
1763 EA AD C0 LD (C0ADh),A     SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
1766 EA AE C0 LD (C0AEh),A     SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
1769 EA AF C0 LD (C0AFh),A     SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=02 LY=01 LYC=00
176C E0 47    LD (FF00+47h),A  SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
176E E0 48    LD (FF00+48h),A  SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
1770 E0 49    LD (FF00+49h),A  SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
1772 3C       INC A            SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=80 -Z--- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
1773 E0 8D    LD (FF00+8Dh),A  SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
1775 C9       RET              SP=DFFD.0163 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
0163 CD A9 05 CALL 05A9h       SP=DFFF.FFFF BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05A9 F0 FF    LD A,(FF00+FFh)  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05AB E0 8E    LD (FF00+8Eh),A  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05AD CB 87    RES 0,A          SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05AF E0 FF    LD (FF00+FFh),A  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B1 F0 44    LD A,(FF00+44h)  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=00 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B3 FE 91    CP 91h           SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=00 ----- IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B5 20 FA    JR NZ,-6         SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=03 LY=01 LYC=00
05B1 F0 44    LD A,(FF00+44h)  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=00 LY=01 LYC=00
05B3 FE 91    CP 91h           SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=00 LY=01 LYC=00
05B5 20 FA    JR NZ,-6         SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=00 LY=01 LYC=00
05B1 F0 44    LD A,(FF00+44h)  SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=00 LY=01 LYC=00
05B3 FE 91    CP 91h           SP=DFFD.0166 BC=0013.CB.00 DE=00D8.00 HL=014D.20 A=01 F=50 --N-C IE=00 IF=00 LCDC=91 STAT=00 LY=01 LYC=00
```

FantasyGmm, 2022-08-23:
The SEGA emulator is now working properly, only the GB emulator is left and I have no idea what is wrong with it

SendToSouthEast, 2022-10-03:
> 

Did you finish it? Can it build on IDF 4.4.1 now? i also want to post it to esp32c3.

SendToSouthEast, 2022-10-03:
> The SEGA emulator is now working properly, only the GB emulator is left and I have no idea what is wrong with it

Did you finish it? Can it build on IDF 4.4.1 now? i also want to post it to esp32c3.

pebri86, 2022-10-08:
> > > The SEGA emulator is now working properly, only the GB emulator is left and I have no idea what is wrong with it
> > 
> > Did you finish it? Can it build on IDF 4.4.1 now? i also want to post it to esp32c3.
> 
> i build it on idf4.4.2,gb emu still crash

Maybe it related to the esp32s2, regular esp32 only successfully build using idf 4.2, using latest esp-idf would be crash conflict about dma memory for i2s.

closed 2022-10-13 by FantasyGmm