pebri86/esplay-retro-emulation #31: how to add video output

new-soul, 2022-09-16:
I know that there was a module name fabgl can realize the goal.
I found the other device was out via vga.

pebri86, 2022-09-19:
You should add the video/vga routine then you could change to render frame to video output rather than to lcd, you can use i2s 1 for rendering video while i2s 0 for sound output

new-soul, 2022-09-19:
> You should add the video/vga routine then you could change to render frame to video output rather than to lcd, you can use i2s 1 for rendering video while i2s 0 for sound output

Is it colorful? I found it was black and white from other website.

pebri86, 2022-09-22:
> > You should add the video/vga routine then you could change to render frame to video output rather than to lcd, you can use i2s 1 for rendering video while i2s 0 for sound output
> 
> Is it colorful? I found it was black and white from other website.

The implementation is depend on you it could be colorful or b/w it should be possible

closed 2024-12-22 by pebri86