pebri86/esplay-retro-emulation #36: Application fails to start

haleyxu, 2023-01-14:
Hi, I recently made this fun project, but I'm running into a problem. The base firmware works fine, but when I select esplay-retro-emu. ESPlay will be black screen, reboot. I wonder if this is caused by the difference in Flash space.
Below is my configuration:
ESP32-WOROVER-E 8MB PSRAM + 8MB Flash

pebri86, 2023-01-17:
Try to compile from source and reconfig according to your esp32 module.

haleyxu, 2023-01-17:
thanks，I have solved the question

---- Replied Message ----
| From | Fuji ***@***.***> |
| Date | 01/17/2023 10:38 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [pebri86/esplay-retro-emulation] Application fails to start (Issue #36) |

Try to compile from source and reconfig according to your esp32 module.

closed 2024-12-22 by pebri86